ssec-jhu/evolver-ng #67: Global evolver instance at the app layer

jamienoss, 2024-05-09:
Currently, we have:
```python
# evolver.app.main

app = FastAPI()
evolver = Evolver()
```

This won't work when running the app from uvicorn using multiple workers, they'll each get their own copy that won't be the same instance. It's possible that this may not matter, however, I expect that multiple serial connections will not work.

For the app on the box, we probably don't want to run with multiple workers anyhow.

Running the app elsewhere though, should still be ok as this would have to go indirectly using the web app connection rather than the serial connection.

From my understanding either way, the more correct FastAPI way to do this, is to stash it in ``app``, though I'm not yet sure of what happens there.
```python
# evolver.app.main

app = FastAPI()
app.evolver = Evolver()
```

amitschang, 2024-05-24:
I agree that there is no case for running multiple workers, there should only be a single evolver (manager) process due to both the serial and control functionality to not conflict.

I don't quite understand the latter part. What would be the reason the evolver instance should be stored in a property of the app? I doubt that property would ever end up being something used in fastapi internals (but could, I suppose), also wondering what problems it would solve?

jamienoss, 2024-05-28:
Global scope just isn't sufficient here, for example, if ``evolver.app.main.evolver`` was ever nullified (for whatever reason), ``from evolver.app.main import evolver`` wouldn't work as this refs the state of ``evolver`` at import time, not when it was updated (nullified) as it's not a pointer.

A more obvious scenario is that addressed in #76  where it's more correct to instantiate the instance of evolver in the startup event, in which case it's not even present at import time.

It also just makes it easier to assign to rather than having to do something like ``globals()["evolver"] = ...`` from path funcs in ``evolver.app.main``.

jamienoss, 2024-05-30:
See https://github.com/ssec-jhu/evolver-ng/pull/73#discussion_r1619464865
>I see now what is going on here. I think the correct location is under `app.state` (e.g. `app.state.evolver`), which is a starlette `State` datastructure (https://github.com/encode/starlette/blob/master/starlette/datastructures.py#L680) and is meant for arbitrary app/request state. This way it avoids potential to conflict with app internals

PR #76 (2024-06-03): Post review changes to #74
Improvements to #74
Resolves #67 

 - Adds ``save`` and ``load`` methods to ``evolver.base.BaseConfig``
 - Separate settings for application settings.
 - Reading the docs, it's now better to use a lifespan event rather than individual startup and shutdown events.
 - Instantiate app ``evolver`` instance on startup, not at module level. (This required #67 to be resolved - see https://github.com/ssec-jhu/evolver-ng/issues/67#issuecomment-2135687714)

See https://github.com/ssec-jhu/evolver-ng/pull/74#issuecomment-2135456108
>I think the files should rep descriptors not  [... cut]
PR #73 (2024-06-06): Refactor usage of creation pattern.
Resolves #53
Resolves #67

# Changes:
 - This propagates the constructor pattern added in #37 to all other classes.
   - adds full init sigs to classes rather than just passing and storing a config.
   - unpacks config to actual instance attributes, i.e., internally stuff isn't ref'd by ``self.config.attr`` but ``self.attr``. 
   -  add ``*args``, ``**kwargs``, and call to ``super().__init__()`` to allow for inheritance.
 - Some extensions to that laid out in #37 have been added.
   - Ease of conversion between ``BaseConfig``, ``ConfigDescriptor``, and ``Bas [... cut]
closed 2024-06-06 by jamienoss