ssec-jhu/evolver-ng #82: interface logging handler duplication

amitschang, 2024-06-06:
The base interface does some logging setup, but in case there are two things of the same type or name, we can get duplicates of the handler, for example, if `MySensor` is descendent of `BaseInterface`

```py
x = MySensor()
y = MySensor()
print(x.logger.handlers)
print(y.logger.handlers)
```

what we see is:
```py
[<StreamHandler stderr (INFO)>, <StreamHandler stderr (INFO)>]
[<StreamHandler stderr (INFO)>, <StreamHandler stderr (INFO)>]
```

because a reconfigure creates new instances in a running process (so that the logger sticks around) , this will likely lead to a lot of increasingly duplicate logs.

It's nice to have the `logger` in there, but I suspect that most cases we'll want to set up the handlers/formatters in root a certain way and have the loggers derive from there, and otherwise they should probably be set explicitly by the user - so the `BaseInterface` could prob refrain from setting the handler at all.

jamienoss, 2024-06-06:
Good catch! This can be handled by adding the appropriate code to remove the logger in ``BaseInterface.__del__``.

Want me to quickly handle it?

jamienoss, 2024-06-06:
Actually, I think it's more a question of not adding new handlers after ``getLogger`` etc.

amitschang, 2024-06-06:
yes, I would advocate for not adding handlers in the base

jamienoss, 2024-06-06:
I think it's totally ok to add them in the base, just not if they already exist as desired.

jamienoss, 2024-06-06:
Also, the point of the ``name`` attr is to differentiate between instances.

PR #83 (2024-06-07): Don't duplicate log handlers
Resolves #82

This also moves the log level and format to configurable settings.
commit 83e111a9 (2024-06-07): Don't duplicate log handlers (#83)

* Don't duplicate log handlers for instance of BaseInterface of the same name

Resolves #82

* Post review changes

---------
closed 2024-06-07 by jamienoss