ssec-jhu/evolver-ng #90: vials do not get list from evolver by default

amitschang, 2024-06-17:
When testing on the evolver box, I started with this config:

```yaml
vials:
- 0
- 5
- 10
- 15
hardware:
  od:
    classinfo: evolver.hardware.standard.od_sensor.ODSensor
    config:
      addr: od_90
  temp:
    classinfo: evolver.hardware.standard.temperature.Temperature
    config:
      addr: temp
```

and was expecting (hoping, maybe) that the vials list for sensors would be that of top-level evolver, however inspected state revealed all 16 vials. This could be corrected by specifying the `vials` in the `config` section of each hardware (as below), but this seems confusing and redundant.

Marking this as a bug since I think the desired behavior with the above config is to carry over the top-level vials by default.

----

Appendix:

This config produces desired result:

```yaml
vials:
- 0
- 5
- 10
- 15
hardware:
  od:
    classinfo: evolver.hardware.standard.od_sensor.ODSensor
    config:
      addr: od_90
      vials: [0, 5, 10, 15]
  temp:
    classinfo: evolver.hardware.standard.temperature.Temperature
    config:
      addr: temp
      vials: [0, 5, 10, 15]
```

jamienoss, 2024-06-20:
This is related to https://github.com/ssec-jhu/evolver-ng/issues/85#issuecomment-2159217228 and the conversation we had offline in regard to there currently being no functionality in ``Evolver`` to act as a manager to determine the config defaults or precedence when it's responsible for instantiating hardware, or any similar object with a ``Config``.

jamienoss, 2024-07-31:
@imaitland Suggested a good idea to use a global that pulls from the evolver instance.

This would totally work from the app layer where an instance of evolver exists, but not from the framework/SDK.

A default factory on this field could be added that checks ``app.state.evolver`` for defaults. Maybe, 🤔 as this bleeds the app and the SDK too closely...

jamienoss, 2024-07-31:
FYI this is something that should def be addressed, just a question of how.