ssec-jhu/evolver-ng #140: Problem running app according to README Example Run

ezirayw, 2024-08-02:
Tried to run the application locally using the command `tox -e test exec -- python -m evolver.app.main`.  The following error pops up:

> ERROR:    Traceback (most recent call last):
  File "/Users/ezirayimerwolle/Documents/GitHub.tmp/Khalil_Lab/evolver-ng/.tox/test/lib/python3.11/site-packages/starlette/routing.py", line 732, in lifespan
    async with self.lifespan_context(app) as maybe_state:
  File "/Users/ezirayimerwolle/.pyenv/versions/3.11.9/lib/python3.11/contextlib.py", line 210, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ezirayimerwolle/Documents/GitHub.tmp/Khalil_Lab/evolver-ng/evolver/app/main.py", line 21, in lifespan
    app.state.evolver = Evolver.create(Evolver.Config.load(app_settings.CONFIG_FILE))
                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ezirayimerwolle/Documents/GitHub.tmp/Khalil_Lab/evolver-ng/evolver/base.py", line 75, in load
    with open(file_path, encoding=encoding) as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'evolver.yml'

This issue only started recently, so I imagine there is a recent commit that either broke some sort of logic or the file is no longer being checked in.

ezirayw, 2024-08-02:
Application runs after making changing the following to false: 
https://github.com/ssec-jhu/evolver-ng/blob/782ee1f0bb2675857bcee4664fe21e4eb61d242b/evolver/settings.py#L15

However, browser cannot connect to port 8000 to check API documentation, which I imagine is in the evolver.yml file.

jamienoss, 2024-08-02:
@ezirayw The default port is ``8080``, try that.
See https://github.com/ssec-jhu/evolver-ng/blob/main/evolver/settings.py#L22

amitschang, 2024-08-03:
We should also update the README so it works out-of-the-box

amitschang, 2024-10-09:
I think with the above issue this is completed (along with #142)

closed 2024-10-09 by amitschang