ssec-jhu/evolver-ng #153: Vial physical layout specification in evolver

amitschang, 2024-08-22:
Right now we have a simple list of the indexes of vials, but this doesn't give enough information about there physical layout unless one assumes the default 16 vial 4x4 grid. 

We should support a configuration that enables both an understanding of the layout of available slots and to specify which are occupied, possibly in a way where the user can assign "names" as needed.

For example:

```
[[0, None, None, None],
 [None, 5, None, None],
 [None, None, 10, None],
 [None, None, None, 15]]
```

or possibly (assuming the numbers don't have to map strictly to top-down-left-right order:

```
[[0, None, None, None],
 [None, 1, None, None],
 [None, None, 2, None],
 [None, None, None, 3]]
```

Assuming we have a 4 x 4 grid but only inhabit one diagonal.

2-d assumption seems appropriate, if needed

jamienoss, 2024-08-27:
```[[0, None, None, None],
 [None, 5, None, None],
 [None, None, 10, None],
 [None, None, None, 15]]
```
With the above the 5, 10, etc are somewhat redundant and would also require validation that someone didn't do:
```
[[0, None, None, None],
 [None, 6, None, None],
 [None, None, 3, None],
 [None, None, None, 8]]
```

I think a concrete approach might offer the least confusion - give each vial a physical number on the box and have the vial ID (int) encode this. This could be as simple as a pic of the box with each vial labeled/numbered. Then they just become vial IDs 0-15. The alternative would have to use a mapping of vial ID to serial address (which is currently just the ordinal, i.e., 0-15).

jamienoss, 2024-08-28:
Yep that seems cool! Though, I think making ``vial_index_order`` configurable will give us a headache in having to constantly account for different orientations. I would just make this static and add it to the docs etc.

jamienoss, 2024-08-28:
Existing implementations exist for this kind of thing, e.g., subplot layouts, my vote would be to role with one of those, i.e., ``[1, 2]`` notation as opposed to ``[2]``. This way the UI can render it accurately as the user has it laid out on the bench in front of them.

amitschang, 2025-01-10:
I'll work on this. Current thought is that we don't need to do a huge amount here in the backend, just provide the config option that can then be consumed in the frontend for visualization. I think we can probably add

```
vial_layout  = [x, y][[, z]]
```

to support grids up to 3d, and just say that it must be left-to-right-bottom-to-top-bottom-to-top order. `vial` can stay since it represents from that grid which ones are plugged in

PR #243 (2025-01-28): vial layout
This adds configuration and validation for a grid layout specification. I'm not sure at the moment how else this needs to be presented or used in the backend, so it is just the configuration for now, which could be used in frontend for presentation.

resolves #153.
closed 2025-01-28 by amitschang