ssec-jhu/evolver-ng #165: Store measured data from calibration in Calibration.Config

jamienoss, 2024-09-04:
Punt from #157 

See https://github.com/ssec-jhu/evolver-ng/pull/157#issuecomment-2327255025

```python
        measured_data: dict | None = Field(
            None,
            description="The data measured by a calibration procedure run. This can be specified instead of specifying "
            "some transformer config params, e.g., polynomial coefficients. This data will be fitted upon "
            "first use.",
        )
```

imaitland, 2024-10-10:
Worked this into the calibration procedure pr with Jamie's feedback.

amitschang, 2025-03-03:
we store the measured data in calibration state model object, typically written via procedure. See also #253 and #268

closed 2025-03-03 by amitschang