ssec-jhu/evolver-ng #183: Resolve latency issue in history endpoint

amitschang, 2024-10-09:
The standard history implementation is file-backed, which has a lot of nice features but even with the time-based partitioning it is noticeably slow on the raspberry pi.

Maybe storing a certain window in-memory, finding a format more suitable, or maybe just tuning partition size, etc might help.

jamienoss, 2024-10-10:
Perhaps we just need a newer/better SD card. It shouldn't be slow.
How slow is it, and how much was being written for you to notice this?

amitschang, 2024-10-10:
It is a brand new SD card. I think one thing to blame is that there is a scan to get schema, but this should not be necessary. First shot could be to remove the need for that, then keep a few hours or so in memory, or similar

PR #190 (2024-10-15): [history] Add in-memory buffer of configurable number of partitions
This is second of set of changes to make history perform better. For recent data (which should be most common to query) we additionaly store in-memory and query that if possible. Data are always also written out to disk and queryable when not satisfied in memory buffer.

brings UI latency from a couple seconds down to 10s of milliseconds for default query.

Resolves #183
closed 2024-10-15 by amitschang