ssec-jhu/evolver-ng #193: Make Transformer parameters optional and fallback to Calibrator.measured data to then gen transformer params

jamienoss, 2024-10-16:
We want Calibrators and Transformers to be trivially configurable and thus swappable. To do this, we need to make ``Transfomer.Config.parameters`` optional such that only some config fields are populated, e.g., polyfit degree, but not the actual polynomial coefficients.

Then, a calibrator would need ``measured_data`` (relates to #165) to call ``Transformer.fit`` to fit the measured data and generate the desired transformer parameters.

If both transformer optional params and calibrator optional measured data aren't present, obviously raise.

jamienoss, 2024-10-16:
@imaitland FYI

amitschang, 2025-03-03:
We have the tools now for measured data to be stored and to do refit on initialization and during procedure using classes provided. See #268 and references therein

closed 2025-03-03 by amitschang