ssec-jhu/evolver-ng #208: Marking which controllers are active on which vials

amitschang, 2024-11-13:


imaitland, 2024-12-17:
I presume the UI part of this is blocked/needs some kind of controllers/experiments api.

amitschang, 2024-12-17:
> I presume the UI part of this is blocked/needs some kind of controllers/experiments api.

I think the controllers config should support a vial list so `Controller.Config.vials` could be inspected to see which vials it ought to be active on (or absent would be all of them)

In reality - to be precise - it might be more nuanced than that since it could be a controller that only sends control commands to a hardware which itself is active on only certain vials, but we might simply start with the vials config.

imaitland, 2024-12-17:
Yeah i think the best place for that is on the backend.

e.g. 

get `/controllers` endpoint accepts a `vials` query param and returns an object, indexed by vial with the active controllers on that vial...or similar.

This lets us in the backend encode whatever rules may exist for a controller applying to all vials, or a particular vial, or a sub-set of vials.

Currently as you say, simply "just look at what vials a controller refs in its config", later ... whatever else...

IMO this is preferable to the front end doing this ad-hoc basis, e.g. the answer to the question "does a controller with no vials in its config apply to all vials, no vials, some vials subject to some intermediate condition?" should be encoded in the backend as it is an assumption that exists across the system.

amitschang, 2024-12-17:
Makes sense, we can add a getter to the manager and corresponding web api as you suggest. It might be nice as well having a get `/controller/{controller}` that returns the controller config plus recent logs in one call, for when drilling down into the list returned by `/controllers`.

amitschang, 2024-12-17:
made #233

closed 2026-05-11 by ryanhausen