ssec-jhu/evolver-ng #213: Bug in nested rebuilding of Config as ConfigDescriptor

jamienoss, 2024-11-21:


amitschang, 2024-11-25:
haha, github doesn't understand the "not" in from of "resolve"!

amitschang, 2025-04-17:
I believe this is resolved by #295, the above example:

```py
from evolver.device import Evolver
from evolver.hardware.standard.temperature import Temperature
from evolver.calibration.standard.calibrators.temperature import TemperatureCalibrator

#calibrator=None
calibrator = TemperatureCalibrator()
x = Evolver(hardware={'test': Temperature(addr='x', calibrator=calibrator)}).config_model
print(x)
```

yielding the expected structure:

```
name='Evolver' namespace='unspecified' vial_layout=[4, 4] vials=[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15] hardware={'test': ConfigDescriptor(classinfo=<class 'evolver.hardware.standard.temperature.Temperature'>, config={'name': 'Temperature', 'calibrator': {'classinfo': 'evolver.calibration.standard.calibrators.temperature.TemperatureCalibrator', 'config': {'name': 'TemperatureCalibrator', 'dir': 'calibration_files', 'created': '2025-04-17T18:38:45.539462', 'expire': 'P168D', 'input_transformer': None, 'output_transformer': None, 'no_refit': False, 'calibration_file': None, 'procedure_file': None, 'vials': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15], 'default_input_transformer': None, 'default_output_transformer': None}}, 'vials': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15], 'addr': 'x', 'slots': 16, 'serial_conn': None})} experiments={} serial=ConfigDescriptor(classinfo=<class 'evolver.serial.EvolverSerialUART'>, config={'name': 'EvolverSerialUART', 'port': '/dev/ttyAMA0', 'baudrate': 9600, 'timeout': 1.0}) history=ConfigDescriptor(classinfo=<class 'evolver.history.standard.HistoryServer'>, config={'name': 'HistoryServer', 'namespace': None, 'partition_seconds': 3600, 'buffer_partitions': 3, 'default_window': 3600, 'default_n_max': 5000}) enable_control=True interval=20 raise_loop_exceptions=False abort_on_control_errors=False abort_on_commit_errors=False skip_control_on_read_failure=True log_level=21
```

and with pydantic upgraded to the latest version

PR #216 (2024-11-25): Add nested config test and temp downgrade of pydantic
Relates to #213

This still needs further investigation to determine whether something will remain broken for newer versions of ``pydantic``.  This does not resolve #213, but will temporarily band-aid it.
closed 2025-04-17 by amitschang