ssec-jhu/evolver-ng #235: Bad calibration result crashes application

amitschang, 2024-12-18:
Using the default calibration for OD90 seems to have parameters incompatible with our sensors, in that for some vials nan is produced, and this then results in crashing the application (not just exception isolated to endpoint), with error like (snipped for brevity):

```
Dec 18 20:40:08 raspberrypi python[857893]: ERROR:    Exception in ASGI application                                                                                          
Dec 18 20:40:08 raspberrypi python[857893]: Traceback (most recent call last):                                                                                               
Dec 18 20:40:08 raspberrypi python[857893]:   File "/home/pi/evolver-venv/lib/python3.11/site-packages/uvicorn/protocols/http/httptools_impl.py", line 399, in run_asgi      
<<snipped>>
Dec 18 20:40:08 raspberrypi python[857893]:   File "/home/pi/evolver-venv/lib/python3.11/site-packages/starlette/responses.py", line 187, in render
Dec 18 20:40:08 raspberrypi python[857893]:     return json.dumps(
Dec 18 20:40:08 raspberrypi python[857893]:            ^^^^^^^^^^^
Dec 18 20:40:08 raspberrypi python[857893]:   File "/usr/lib/python3.11/json/__init__.py", line 238, in dumps
Dec 18 20:40:08 raspberrypi python[857893]:     **kw).encode(obj)
Dec 18 20:40:08 raspberrypi python[857893]:           ^^^^^^^^^^^
Dec 18 20:40:08 raspberrypi python[857893]:   File "/usr/lib/python3.11/json/encoder.py", line 200, in encode
Dec 18 20:40:08 raspberrypi python[857893]:     chunks = self.iterencode(o, _one_shot=True)
Dec 18 20:40:08 raspberrypi python[857893]:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Dec 18 20:40:08 raspberrypi python[857893]:   File "/usr/lib/python3.11/json/encoder.py", line 258, in iterencode
Dec 18 20:40:08 raspberrypi python[857893]:     return _iterencode(o, 0)
Dec 18 20:40:08 raspberrypi python[857893]:            ^^^^^^^^^^^^^^^^^
Dec 18 20:40:08 raspberrypi python[857893]: ValueError: Out of range float values are not JSON compliant
```

Using the default calibration file and values like those read on the device, I can confirm that the fit function produces a `NaN`.  While this is probably the correct thing to do in such a case, we should ensure that the app can handle or gracefully error with such values. 

Oddly, just doing this on python prompt produces non-error looking results:

```py
>>> json.dumps(np.nan)
'NaN'   
```

and 

```py
>>> class X(pydantic.BaseModel):
...   x: float
>>> X(x=np.nan).model_dump_json()
'{"x":null}'
```

So there may be something fastapi specific going on here...

jamienoss, 2024-12-19:
It shouldn't "crash" the application, does it?

amitschang, 2024-12-19:
True, I'm not sure why I wrote that. The process is running but the UI is rendered broken

[image]

jamienoss, 2024-12-19:
Ok, that's ok, I would state the same thing for this. The application should still be usable. We prob just need to add/activate the error handlers and/or better invocation of ``uvicorn`` so that it auto restarts.

amitschang, 2024-12-19:
Shouldn't we fix so that the `NaN`s end up serialized as `nulls`? Or otherwise in a way where the state endpoint can still be called?

jamienoss, 2024-12-19:
Sure, but isn't this ticket 2 in 1? One issue being the ``NaN`` and the other the lack of fault tolerance on the app, right?

amitschang, 2024-12-19:
Found this: https://github.com/fastapi/fastapi/issues/459, looks like it is similar issue. I could reproduce in a test case and see it comes from code in JSONEncoder:

```py
            if not allow_nan:
                raise ValueError(
                    "Out of range float values are not JSON compliant: " +
                    repr(o))
```

default must be to allow_nan=False (doesn't look like it is specifcally stated in fastapi).

---
And no, I think it is just the one issue. Sorry for the confusion, UI seems to be more sensitive than it needs (for instance the "online" indicator should only hit `/healthz`, but maybe does more)

jamienoss, 2024-12-19:
>And no, I think it is just the one issue. Sorry for the confusion, UI seems to be more sensitive than it needs (for instance the "online" indicator should only hit /healthz, but maybe does more)

Ah ok, cool.

@imaitland can yo take a look and see what the issue there might be, please?

PR #237 (2025-01-07): switch json response handler to enable NaN in serialization
FastAPI provides orjson and this looks to be a way that we can allow `NaN` in the json responses. Given passing tests we have some confidence that it produces correct json in other cases.

Resolves #235
closed 2025-01-07 by amitschang