ssec-jhu/evolver-ng #247: Bad classinfo creates Config object instead of failing

amitschang, 2025-01-23:
In debugging some new code that uses evolver `BaseConfig` with `ConfigDescriptors` as a type within the model, when passing in a `classinfo` that has a typo (and thus does not reference a class that can be imported), the system replaces the object with a plain `Config` object instead of failing.

Ex:

```py
class Experiment(BaseConfig):
    controllers: list[ConfigDescriptor | Controller]
```

and putting in:
```py
{"controllers": [ {"classinfo": "typo", "config": {}} ]}
```
  
caused the error

```
s]
                                         ^^^^^
  File "/Users/arik/ws/ssec/evolver-ng/.tox/dev/lib/python3.11/site-packages/pydantic/main.py", line 811, in __getattr__
    raise AttributeError(f'{type(self).__name__!r} object has no attribute {item!r}')
AttributeError: 'Config' object has no attribute 'run'
```

Showing that what is passed through ends up as a `Config` object. When the typo is corrected, the application runs as expected.

**NB**: This seems to only happen if the type hint is `ConfigDescriptor | Controller`, if it is instead simply `ConfigDescriptor` it winds up erroring about the module.

imaitland, 2025-04-09:
Any `config: null` will cause this issue.

e.g.

[image]

imaitland, 2025-04-09:
```
config: {}
```
also ok...

imaitland, 2025-04-09:
[image]

jamienoss, 2025-04-09:
@amitschang & @imaitland this is a similar, if not duplicate, issue of #213.

ezirayw, 2025-04-09:
went through #213 and #216 and saw that potentially putting a ceiling on the pydantic version helped fix problems. not sure if its a bandaid for now but ideally would want to prioritize solutions that work off of the current pydantic version to avoid future code-breaking issues

amitschang, 2025-04-09:
I think I found some leads here. There are two things going on:

1. For union types, pydantic is trying to validate types and return the first one that properly validates, so when ConfigDescriptor fails (e.g. due to bad import, which in this case is a validation error) it moves on to the other kind (e.g. `Controller`)
1. When it tries to validate a type that inherits from `BaseInterface`, the schema it gets back:
   *  is that of the `Config` object on it (due to the overload of `__get_pydantic_core_schema_` at https://github.com/ssec-jhu/evolver-ng/blob/main/evolver/base.py#L381)
   * Seems to validate any input data, for example `Controller.Config.model_validate_json('{"classinfo": "abcd"}')` returns a `Config` object without error

For (1), we can write a wrapping validator that essentially duck-types for Config-descriptor  - if it has "classinfo" field, fail fully if we cannot properly create a ConfigDescriptor. I'm not totally sure about (2), we don't really want to arrive at the `Config` object here at all. Perhaps the base config objects need an after validator to ensure the types are as expected

jamienoss, 2025-04-09:
>For (1), we can write a wrapping validator that essentially duck-types for Config-descriptor - if it has "classinfo" field, fail fully if we cannot properly create a ConfigDescriptor.

Is that not the same as just dropping the union? E.g., ``Controller`` doesn't have a ``classinfo`` so will always fail. I'd have to refresh my memory, but the crux of the issue is how do you tell the difference between a malformed ``ConfigDescriptor`` and a completely valid ``Config`` - I don't believe you can. Besides, isn't ``classinfo`` already a required field, so validation of the kind you suggest above should already be happening.

I believe the correct solution is to add explicit serialization rather than letting pydantic dynamically choose. The simplest solution to this is to drop all type unions, as we have discussed in the past or carry on with #246.

amitschang, 2025-04-10:
Dropping the unions is probably fine, we can also advise to use config descriptor only when referencing things that are not plain models, but union None (optional) is OK.

The union probably spoke more to the expected class of the configdescriptor - can we use generics to provide information on the thing expected?

e.g. the union hint below is implying we want a `Calibrator`

```py
calibrator: ConfigDescriptor | Calibrator
```

but this might be what we are really trying to say:

```py
calibrator: ConfigDescriptor[Calibrator]
```

closed 2025-04-17 by amitschang