ssec-jhu/evolver-ng #254: Reconcile settings.py and evolver.yml

imaitland, 2025-02-11:
It's not clear how the env vars in `settings.py` relate to similar settings in the `evolver.yml` config.

for example settings.py has variables:
```
ROOT_CALIBRATOR_FILE_STORAGE_PATH: Path = Path("calibration_files")
EXPERIMENT_FILE_STORAGE_PATH: Path = Path("experiment_files")
```

while the evolver.yml may have attributes in config like `dir` and `calibration_file`:

```
  temp:
    classinfo: evolver.hardware.standard.temperature.Temperature
    config:
      addr: temp
      calibrator:
        classinfo: evolver.calibration.standard.calibrators.temperature.TemperatureCalibrator
        config:
          calibration_file: foobar2000.yml
          created: '2024-11-20T21:06:13.563442'
          dir: calibration_files
```

If there's a coherent relationship between the two sources of truth it must be expressed in documentation. If it already exists, or is obvious, this issue can be closed with reference to the answer.

imaitland, 2025-02-11:
I'd also suggest in so far as is possible, the `evolver.yml` should be the single source of truth for **all** config... if discussion exists for the decision to bifurcate / reasoning for a `settings.py` and an `evolver.yml` i'd be curious to know / see it.

amitschang, 2025-02-11:
I agree that the config (that in `evolver.yml` or represented via `Evolver.Config`) is the source of truth for all users settings for the business-logic of running application. 

I think the settings should represent only application-wide defaults that would benefit from tweaking in different environments, and for those settings that affect the startup of the application prior to reading in the `Evolver.Config` (which should be thought of as a thing that happens after application bootstrap, not during).

imaitland, 2025-02-11:
ah got you... so for example:
`ROOT_CALIBRATOR_FILE_STORAGE_PATH`
is better described as:
`DEFAULT_ROOT_CALIBRATOR_FILE_STORAGE_PATH`

jamienoss, 2025-02-11:
"Config" is user facing. It encapsulates the hardware setup, the science, experiments, control, etc. The things that the user wants to do with it.

Settings are dev/devops facing. They encapsulate application and deployment settings, e.g., host and port, in ``evolver.settings.AppSettings``. They also include some defaults, behavior defaults, and things like feature switches, i.e., that in ``evolver.settings.Settings.

They shouldn't and can't be merged.

imaitland, 2025-02-11:
hmmm, yeah i get that, just was curious about where evolver.yml and settings.py overlap. Seems the answer is "they dont" just the naming of some of the settings.py flags are a little confusing.

closed 2025-02-11 by imaitland

jamienoss, 2025-02-11:
> was curious about where evolver.yml and settings.py overlap. Seems the answer is "they dont" just the naming of some of the settings.py flags are a little confusing.

Correct. Overlapping encapsulations should be avoided by definition.